RobotWebTools/ros3djs #640: LaserScan points rotate with the robot

fuuuuuk, 2023-10-25:
**Description**
When my robot undergoes translation or rotation, the radar point cloud also experiences translation or rotation, which is particularly noticeable during rotations,When the robot comes to a standstill, the radar point cloud returns to its normal state.
My understanding is that the tfclient performs coordinate transformations, affecting the translation and rotation of a 3D object, which in turn impacts the points within the object. When a new radar point cloud frame hasn't been rendered yet, the old point cloud frame is shifted due to the object's rotation and translation. Is there a way to synchronously subscribe to both the radar topic message and the tf topic message, ensuring that radar messages correspond correctly to tf messages? Alternatively, can the radar point cloud remain stationary during TF transformations of the object?

- Library Version:ros3djs
- ROS Version:melodic
- Platform / OS:ubuntu

**Steps To Reproduce**

                this.tfClient = new ROSLIB.TFClient({
                                    ros : this.globalData.ros,
                                    fixedFrame :'/map',
                                    angularThres : 0.01,
                                    transThres : 0.01,
                                    rate : 10.0
                                })
                
                this.viewer = new ROS3D.Viewer({
                        divID: 'map',
                        width: 750,
                        height: 750,
                        antialias: true,
                    });
                this.laserclient = new ROS3D.LaserScan({
                    ros: this.globalData.ros,
                    topic: '/scan',
                    tfClient: this.tfClient,
                    rootObject: this.viewer.scene,
                    material: { size: 0.1, color: 0xFF0000 }
                })
                
                this.gridClient = new ROS3D.OccupancyGridClient({
                    ros: this.globalData.ros, 
                    rootObject: this.viewer.scene,
                    continuous: true,
                });
                var arrowNode = new ROS3D.SceneNode({
                    tfClient : this.tfClient,
                    frameID  : '/base_link',
                    object   : new ROS3D.Arrow({
                                    length: 0.5,
                                    headLength:0.4,
                                    shaftDiameter:0.1,
                                    headDiameter:0.3,
                                    color: 0xFD8600
                                }),        
                });
                // console.log(arrowNode)
                this.viewer.scene.add(arrowNode);

**Expected Behavior**

**Actual Behavior**
[image]
[image]

MatthijsBurgh, 2023-10-25:
When the points eventually do converge to the correct position, it sounds like a delay issue. So the library is correct, but it is not fast enough in combination with your hardware.

fuuuuuk, 2023-10-26:
> When the points eventually do converge to the correct position, it sounds like a delay issue. So the library is correct, but it is not fast enough in combination with your hardware.

thank you for your response. I understand that this is due to the delay in transmitting and rendering radar data, which results in the next frame not being rendered immediately after the previous one. During this intermediate phase, a TF transformation occurs, causing the radar point cloud from the previous frame to be translated or rotated along with it. However, my radar operates at a frequency of 5 Hz, and there is inherent delay in rendering radar data that cannot be changed.
When the delay in transmitting or processing radar data is minimal, this offset phenomenon is relatively minor, but still unavoidable.
[image]