RobotWebTools/ros3djs #657: Adding decay time to PointCloud2

Pandananana, 2024-03-08:
It would be very valuable for me if it was possible to have the PointCloud2 stay visible for a set amount of seconds, so that I can easily see what I have scanned already, and to see if overlap is too much or too little. 

This PR is relevant, but is flawed: https://github.com/RobotWebTools/ros3djs/pull/357

If there is an easier / better way that doesn't require modifying ROS3Djs I would also like to know.

MatthijsBurgh, 2024-03-08:
I don't have the time to implement this. In case you write the code, I am happy to review it.

Pandananana, 2024-03-08:
Ok I will look into it. I am guessing it should include a unit test, as mentioned in the other PR?

MatthijsBurgh, 2024-03-08:
That is indeed very desired to keep it maintainable in the future.

angxuejian, 2025-05-12:
Looking forward to this feature, may I ask what the progress is now?