RobotWebTools/ros3djs #658: ROS3Djs in combination with ROS2

davidDS96, 2024-03-12:
Hello all,

I was wondering if the ROS3Djs framework is also supported for usage with ROS2.

Anyone some news on that?

Best regards

MatthijsBurgh, 2024-03-12:
It depends on the support by roslib(https://github.com/RobotWebTools/roslibjs). The currently released version does support all the ROS2 interfaces except actions.

davidDS96, 2024-03-15:
We were testing if we could visualise a PointCloud2 message in the browser using ros3Djs.
We have verified that the PointCloud message arrives properly. However, while the 3D canvas is displayed, no PointCloud is visible. Looking at the errors, we suspect that the problem comes from the tf2_web_republisher package:

```
[rosbridge_websocket-1] [INFO] [1710498853.867756324] [rikkert.rosbridge_websocket]: Client connected. 1 clients total.
[rosbridge_websocket-1] [ERROR] [1710498853.879284256] [rikkert.rosbridge_websocket]: [Client 91d04f66-b86c-4754-8333-0fc7748421c6] [id: advertise:/tf2_web_republisher/goal:1] advertise: Unable to import msg class TFSubscriptionActionGoal from package tf2_web_republisher. Caused by module 'tf2_web_republisher.msg' has no attribute 'TFSubscriptionActionGoal'
[rosbridge_websocket-1] [ERROR] [1710498853.880151856] [rikkert.rosbridge_websocket]: [Client 91d04f66-b86c-4754-8333-0fc7748421c6] [id: subscribe:/tf2_web_republisher/feedback:3] subscribe: Unable to import msg class TFSubscriptionActionFeedback from package tf2_web_republisher. Caused by module 'tf2_web_republisher.msg' has no attribute 'TFSubscriptionActionFeedback'
[rosbridge_websocket-1] [INFO] [1710498853.881981997] [rikkert.rosbridge_websocket]: [Client 91d04f66-b86c-4754-8333-0fc7748421c6] Subscribed to /rs_bpearl/pointcloud
[rosbridge_websocket-1] [ERROR] [1710498854.026874867] [rikkert.rosbridge_websocket]: [Client 91d04f66-b86c-4754-8333-0fc7748421c6] [id: publish:/tf2_web_republisher/goal:5] publish: Cannot infer topic type for topic /tf2_web_republisher/goal as it is not yet advertised
```
Do you have any suggestions on how to fix this problem?

MatthijsBurgh, 2024-03-15:
That is not compatible with ROS2 (yet)

davidDS96, 2024-03-15:
Ok, that's too bad.
Is there any indication when this comes available?

MatthijsBurgh, 2024-03-15:
The repo is actually archived. So nobody is maintaining it for now. So I think you need to refactor it for ROS2. But than you also need an unreleased version of roslibjs, which does support ROS2 actions.

davidDS96, 2024-03-15:
Ok. 
I see in the roslibjs repo (https://github.com/RobotWebTools/roslibjs/tree/develop/examples) that some examples are added 2 weeks ago with regards to ROS2 and action support. However, tf and urdf have not been updated in the past 10 months? Is there any chance an update of these will occur?

MatthijsBurgh, 2024-03-16:
I can't say. You need to ask @EzraBrooks and @sea-bass .

sea-bass, 2024-03-16:
We are possibly working on porting the tf2_web_republisher soon. But this is a set of open source repos, so anyone is always welcome to contribute!

davidDS96, 2024-03-29:
I found the following repository (https://github.com/Wisc-HCI/tf2_web_republisher_py) containing a port of the ```tf2_web_republisher``` for ROS2. Does anyone have any experience using this package?

sea-bass, 2024-03-29:
I can only speak to this fork which I've seen work: https://github.com/pac48/tf2_web_republisher

davidDS96, 2024-04-11:
Hi everyone!

After some busy days, I've found the time to test the ```tf2_web_republisher``` [fork](https://github.com/pac48/tf2_web_republisher) for visualising a 3D pointcloud using ROS3Djs.

To visualise a 3D pointcloud in the browser, I've used the following Javascript:

```javascript
<script type="text/javascript" type="text/javascript">

    const ros = new ROSLIB.Ros({
        url: 'ws://192.168.69.20:9090'
    })

    // Create the main viewer.
    const viewer = new ROS3D.Viewer({
    divID : 'viewer',
    width : 800,
    height : 600,
    antialias : true
    });

    // Setup a client to listen to TFs.
    const tfClient = new ROSLIB.TFClient({
        ros : ros,
        angularThres : 0.01,
        transThres : 0.01,
        rate : 10.0
    });

    var cloudClient = new ROS3D.PointCloud2({
        ros: ros,
        tfClient: tfClient,
        rootObject: viewer.scene,
        topic: '/rs_bpearl/pointcloud',
        material: { size: 0.1, color: 0xff00ff }
    });

</script>
```

I have the following terminal output:

```
[rosbridge_websocket-4] [INFO] [1712839238.161715175] [rikkert.rosbridge_websocket]: Rosbridge WebSocket server started on port 9090
[rosbridge_websocket-4] [INFO] [1712839243.652251997] [rikkert.rosbridge_websocket]: Calling services in existing thread
[rosbridge_websocket-4] [INFO] [1712839243.653098323] [rikkert.rosbridge_websocket]: Sending action goals in existing thread
[rosbridge_websocket-4] [INFO] [1712839243.654413931] [rikkert.rosbridge_websocket]: Client connected. 1 clients total.
[rosbridge_websocket-4] [ERROR] [1712839243.688801503] [rikkert.rosbridge_websocket]: [Client 0cbedeb3-be59-4acb-8cec-54be58ccef25] [id: advertise:/tf2_web_republisher/goal:1] advertise: Unable to import tf2_web_republisher.msg from package tf2_web_republisher. Caused by: No module named 'tf2_web_republisher'
[rosbridge_websocket-4] [ERROR] [1712839243.697831669] [rikkert.rosbridge_websocket]: [Client 0cbedeb3-be59-4acb-8cec-54be58ccef25] [id: subscribe:/tf2_web_republisher/feedback:3] subscribe: Unable to import tf2_web_republisher.msg from package tf2_web_republisher. Caused by: No module named 'tf2_web_republisher'
[rosbridge_websocket-4] [INFO] [1712839243.752100615] [rikkert.rosbridge_websocket]: [Client 0cbedeb3-be59-4acb-8cec-54be58ccef25] Subscribed to /rs_bpearl/pointcloud
[rosbridge_websocket-4] [ERROR] [1712839243.913545312] [rikkert.rosbridge_websocket]: [Client 0cbedeb3-be59-4acb-8cec-54be58ccef25] [id: publish:/tf2_web_republisher/goal:5] publish: Cannot infer topic type for topic /tf2_web_republisher/goal as it is not yet advertised
```

and unfortunately no pointcloud is visualised.

pac48, 2024-04-11:
@davidDS96 You will also need this fork on roslibjs https://github.com/pac48/roslibjs/tree/pr-add-TF2Client in order to use the tf2_web_republisher. I added a class called TF2Client that you need to use. Also, make sure you have the rosidl-generator-py package installed.

DeveloperKSH, 2024-04-16:
@pac48  I would appreciate it if you could let me know how to use the rosidl-generator-py package after installing it.

davidDS96, 2024-04-16:
@pac48 I tried to integrate your fork of [roslibjs](https://github.com/pac48/roslibjs/tree/pr-add-TF2Client) into my project but I am running into a few problems with regards to Javascript: several imports are not working or are not found
For instance: 
```js 
import assign from "object-assign"
```
is throwing an error for not finding the package. Unfortunately, this is only one of the error messages I have.

Nick-PAN, 2024-04-20:
@pac48 I would appreciate it if you tell me how to build the roslibjs? I want to use it with ros3djs, will it work?

pac48, 2024-04-21:
@Nick-PAN You should be able to clone the roslibjs repo and run `pnpm install` to install of of its dependencies. Then, you should be able to add it to your project by running `pnpm install /path/to/roslibjs`. That is if you want to make changes to it. Otherwise, you can probably just add it to your `package.json` like this `"github:pac48/roslibjs#pr-add-TF2Client"`.

chanhhoang99, 2024-04-22:
Hello @pac48 , could you share your roslibjs prebuild version ?

Nick-PAN, 2024-05-14:
Thank you @pac48 ! I solved my problem by your reply, and displayed the Universal Robot model on web successfully. 

> @Nick-PAN You should be able to clone the roslibjs repo and run `pnpm install` to install of of its dependencies. Then, you should be able to add it to your project by running `pnpm install [...]

kavikode, 2024-07-03:
how do we install and run ROS3Djs with ROS2 Humble to visualize a URDF and interact with the joint state publisher in the browser because there are no pre-built libraries existing?

ShauryaJain03, 2025-06-16:
@DanieleVeri any updates? Have you figured out how to use ros3djs on ros2 humble?

DanieleVeri, 2025-06-17:
@ShauryaJain03 no, i'm still trying but there is some little progress... i'll let you know

ShauryaJain03, 2025-06-17:
@DanieleVeri great, I'm trying to visualise velodyne VLP 16 LiDAR's point cloud on the browser, just as we would do in rviz2 by subscribing to the /velodyne_points topic and similarly display depth camera and other sensor data on the browser for an interface I am building for one of my robots. Any help would be appreciated :)

ShauryaJain03, 2025-06-17:
@DanieleVeri do let me know if you find any solution to it!