RobotWebTools/ros3djs #671: Change the javascript import

MLDario, 2024-06-11:
Hello all, 
I'm following the tutorial at: http://wiki.ros.org/ros3djs/Tutorials/isualizingAURDF.
I realized that the import links in the javascript file are not working anymore and i solved it using:

<script src="https://cdn.jsdelivr.net/npm/eventemitter2@6.4.9/lib/eventemitter2.min.js"></script>
<script src="https://cdn.jsdelivr.net/npm/roslib@1/build/roslib.min.js"></script>
<script src="https://cdn.jsdelivr.net/npm/three@0.89.0/build/three.min.js"></script>
<script src="https://cdn.jsdelivr.net/npm/ros3d@1.1/build/ros3d.min.js"></script>
<script src="https://robotwebtools.github.io/ros3djs/"></script>

I spent quite a long time finding them, there's no reason why other people should do it too, we should update them.

Thank you :)

MatthijsBurgh, 2024-06-11:
I am not able to create a wiki account, so I can't update them.

The code snippet you provided, is also outdated. Please check the examples for the current `imports`.

MLDario, 2024-06-11:
I don't know why but the ones in the examples don't work for me.

The browser console give the error: Uncaught ReferenceError: ROS3D is not defined

MatthijsBurgh, 2024-06-11:
Those only work when you have the examples in the repo...

So when you need more help, you need to provide more information