RobotWebTools/ros3djs #692: PointCloud2 is not display ﻿

dyjandxy, 2024-09-24:
I have a relatively large point cloud data with max_pts set to a maximum of 150000000. PointCloud2 loads the latest data every time and cleans up the old data. How to make the data load and display every time

MatthijsBurgh, 2024-09-25:
Your message is contradicting. You say it drops the old data and loads the new data. Which is the desired behaviour IMO.
Though you also ask how to load the data and display it.
So what is the real question?

Also share some code if you want people to help you.

Just start by using the [example](https://github.com/RobotWebTools/ros3djs/blob/develop/examples/depthcloud.html). Adapt it to your needs.

dyjandxy, 2024-09-26:
This is the code for me to load point cloud according to the official example
[image]
This is the source code of PointCloud2 that I reviewed. This code updates the latest received data every time, and old data is replaced
[image]
[image]
If I want new data to accumulate and display data without replacing old data, I should need to change the source code of PointCloud2

closed 2024-10-10 by dyjandxy