RobotWebTools/ros3djs #695: Repackaging ros3djs

dyjandxy, 2024-10-10:
Does the repackaging of ros3djs need to be executed in Ubuntu environment? Can't it be done in a Windows environment.

When I repackage ros3djs in the Windows environment, it can be packaged normally, but referencing the packaged ros3djs will result in an error, OcTreeBase is not defined
[image]

MatthijsBurgh, 2024-10-14:
I have no clue what you mean with `repackaging`?

The library is already packaged. See the `build` folder.

MatthijsBurgh, 2024-11-12:
Ah check, you just want to build the code.

I have no clue whether this works in Windows. I also have no clue what you changed. So in your situation, the cause could be Windows, but also your changes. Best test would be try to build the most recent version of the library, without your changes, on Windows.

dyjandxy, 2024-11-12:
I downloaded the latest code in the Windows environment without making any changes. Even after repackaging and compiling the code, the same problem still occurs