RobotWebTools/ros3djs #705: Is ROS3djs Compatable with ROS2 humble

pvg6, 2024-12-08:
**Description**

- Library Version:
- ROS Version: 2 ( humble)
- Platform / OS: Ubuntu22

**Steps To Reproduce**

Is ROS3djs Compatable with ROS2 humble . Is it possible the map and robot movement in web using ros2 humble and ros3djs
**Expected Behavior**

**Actual Behavior**

mattiadutto, 2024-12-11:
I started using ROS3DJS with ROS2 Humble and I'm able to view the map on the web. For the other I'm currently working on it.

3bdul1ah, 2025-02-27:
> started using ROS3DJS with ROS2 Humble and I'm able to view the map on the web. For the other I'm currently working on it.

Started using **ROS3DJS** with **ROS 2 Humble**, and like you, I’m able to view the map on the web. Still working on the other parts.  

Did the same, but **URDF** and **point clouds** require `tf2_web_republisher`. When trying, **rosbridge** throws an error saying it can’t find `TF2ActionFeedback` or something similar.  

Have you managed to get anything working regarding this? Any suggestions for a workaround?

mattiadutto, 2025-02-28:
@3bdul1ah Hi! Since I didn't need the 3D map but just a 2d one, I switched back to ROS2DJS and I was able to publish both the map and the current position of the robot by using AMCL topic subscription.